danomagnum/gologix #2: how to add field in struct by ReadMulti function?

Ri0nXh, 2023-02-17:
Hello, when using the readmulti method, you need to pass a struct, which is identified as a tag by the fields in the struct. If I want to add 5000 fields to this struct in a loop to represent 5000 tags, what is the method?

```go
	type name struct {
		TAG1 float32 `gologix:"TAG1"`
		TAG2 float32 `gologix:"TAG2"`
		TAG3 float32 `gologix:"TAG3"`
                .....
                // Dynamically add 5000 fields
                TAG5000 float32 `gologix:"TAG5000"`
	}
        var n name
        client.ReadMulti(&n)
```
thanks

danomagnum, 2023-02-17:
You can either try generating your struct with [go generate](https://go.dev/blog/generate) or try the new [```ReadList(tags []string, types []CIPType) ([]any, error)```](https://github.com/danomagnum/gologix/blob/2046c13e3916bd61dc905874df20787a5c1d7b5b/read.go#L515) function added in beta 9.

The ReadList function takes a list of tag names and a list of the types for each tag as strings so it should be easy to generate them in a loop.  One downside of this is that the function returns ```[]any``` for the result so you'll have to do type assertion to get the actual values out.

### NOTE:
I haven't done any testing yet on large reads/writes.  I am pretty confident that the read **WILL FAIL** at some size because we will exceed the maximum message size negotiated during connection (the ForwardOpen message sets this to between around 4000 and 500 bytes depending on the device).  It is on the roadmap to take care of converting one read or write function call to multiple messages and merge the results back together seamlessly behind the scenes, but that is not implemented yet.  **So for now you might have to issue multiple reads.**

Ri0nXh, 2023-02-20:
thank you advice, tags can be read.

closed 2023-02-20 by danomagnum