danomagnum/gologix #8: [Bug] Cannot read string using ReadList() / ReadMulti()

charlyjna, 2023-07-05:
Hi,

I am trying to use your driver which is working well, and thank you for your work.

But, I am facing an issue I do not really understand. I cannot read a string using the ReadList() method, or the ReadMulti(). I always get a an error: problem reading tag {TAGNAME}: Struct!.

And if I try to read this same tag using the simple Read() method, it works out-of-the-box.

I tried with CIPTypeString and CIPTypeStruct and get the same error message.

Do you have any idea?

Thanks,

Charly

danomagnum, 2023-07-05:
Interesting. I'll investigate and let you know what I find.

danomagnum, 2023-07-06:
OK, so ReadMulti uses ReadList internally (it parses your struct tags and then generates the arguments for ReadList and uses it to actually get the data then fills the stuct back up).

When you read a STRING type from a logix controller it actually sends back the same response as any other UDT.  The normal Read() method took this into account and handled STRING in a special way.  ReadList() was not doing this extra check to parse the string data out.

I added the check in and added a unit test for this.  Give beta 14 a try when you get a chance and let me know if everything is working or not.

charlyjna, 2023-07-06:
I tried, and it works nice now!
I just had one mistake, but I realized it was because I used _new(string)_ (_*string_) instead of _string_. And it has been identified as an interface{}, so a CIPTypeStruct instead of CIPTypeSTRING.

Thank you very much!