danomagnum/gologix #9: [Question] Timeout

charlyjna, 2023-09-12:
How can I set a connect or read timeout?
Because for now, it can stop after 1 minute or more.

I tried to set SocketTimeout to 1 second, but it panics immediately:
```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x80 pc=0x74d7b2]

goroutine 56 [running]:
github.com/danomagnum/gologix.(*Client).Connect(0xc0002e7b90?)
        /home/charly/go/pkg/mod/github.com/danomagnum/gologix@v0.16.0-!beta/connect.go:14 +0x12
```

I am probably not doing the right thing...

danomagnum, 2023-09-12:
Can you share the code where you're connecting and setting the timeout?

The error code looks like you're getting a nil reference on Line 14 of connect.go but the only thing that could be nil there would be the client object itself which is strange if you were setting the SocketTimeout parameter, since you'd need a valid instance to set that in the first place.

I'll try to run a test this evening and see if I can reproduce it.

danomagnum, 2023-09-13:
I confirmed that the test still passes.

This is the code that sets up a client with a specified timeout:

`client := gologix.NewClient("192.168.2.241")
	client.SocketTimeout = time.Minute
	err := client.Connect()
	if err != nil {
		t.Error(err)
		return
	}
	defer func() {
		err := client.Disconnect()
		if err != nil {
			t.Errorf("problem disconnecting. %v", err)
		}
	}()`

charlyjna, 2023-09-13:
Ok, so first I want to thank you for your answer. It confirms to me I am doing right using SocketTimeout.
And, I want to apologize because I realized that, yesterday, I had some network issues I did not know while I was testing the timeout. 

For the first connection, it failed because of the timeout, and it worked without because of this network issue. Now the network works normally, and everything is fine.

I just added a retry in my code to be sure the first connection worked before to continue.

Sorry, and thank you again.

closed 2023-09-13 by charlyjna