danomagnum/gologix #10: Support batch writing of values？

Ri0nGo, 2023-10-11:
Hi，Does it support batch writing of values?

danomagnum, 2023-10-12:
Yes.  It supports writing multiple values at once with `gologix:"<tagname>"` tagged structs.  It also has the ability to do a map[string]interface{} with the writeDict() function where the keys of the map are the tag paths and the values are the values to write, but this isn't exported for some reason.

I'll try and get that exported for easier use (read already has this exported as the ReadList() function) and get an example made up for it.

In the meantime you should be able to use a tagged struct like I said.

danomagnum, 2023-10-13:
I released beta 17 where I have made the function for writing multiple tags at once external.  See "MultiWrite" in the example folder for what I think you're looking for.

Let me know if you have any issues.

Ri0nGo, 2023-10-13:
That is my want. Thank you.

closed 2023-10-13 by Ri0nGo