danomagnum/gologix #18: client.GetAttrList for Identity Object returns expected results only for L7x controllers

emourzi1, 2024-04-25:
client.GetAttrList for Identity Object in my testing returns expected results only for L7x controllers.

For L8z, EN2T, other in-chassis modules and PowerFlex drives I am only getting 6 bytes back 
In getattr.go add a print statement and it is 0 for everything other than  LOGIX5575
count, err := items[1].Int16() // Count
fmt.Println(count) // prints 0
	
Getting single attributes works as expected. 
Is this a known issue?

danomagnum, 2024-04-25:
So an EN2T returns "Service Not Supported" when you do getattr list.  

[image]

I don't have an L8 to test with, but if you can collect a wireshark trace of a minimum replication I can look at it and see if it is a similar thing or not.

I'll try to make some changes to the GetAttrList() to error with the status code if it is not "OK".  That way it will be more clear what is going on.  To see what the L8 is doing, you can look at the status code in the resphdr value on line 204.  https://github.com/danomagnum/gologix/blob/5dd8562f8248f0df0c16e270115090c04ea1b7a8/getattr.go#L204

I'll also look at updating that struct to use the CIPStatus type instead of just an integer to provide human readable diagnostics.

In the meantime, the value can be manually cast to a CIPStatus:

```
	var resphdr cipAttributeResponseHdr
	items[1].DeSerialize(&resphdr)
	//dat := make([]byte, hdr.Length-26)
	//items[1].DeSerialize(&dat)
	if resphdr.Status != uint16(CIPStatus_OK) {
		return nil, fmt.Errorf("Status not OK: %v", CIPStatus(resphdr.Status))
	}

```

emourzi1, 2024-04-25:
I am not a comms expert and it has been a while since I used wireshark, will try.
Meanwhile, I added the status printout. 
L8 controllers (tested multiple) show CIPStatus_TooMuchData , and all other Rockwell devices tested  return CIPStatus_ServiceNotSupported

emourzi1, 2024-04-25:
I started the other way - commented all three of then, everything is working for my L7 and L8.  I am getting CIP identities through this call.
Will let you know if I am able to find out more about the serialization piece.

danomagnum, 2024-04-26:
I published a new release with the removed extra serializations and improved return error values.

Thanks for the good bug report.

closed 2024-04-26 by danomagnum