danomagnum/gologix #19: Connection reset by peer, using ETHERNET-MODULE

jwalke23, 2024-05-30:
I'm trying to set up communication between a PLC and a linux server on the same subnet. I'm using the example [examples/Server_Class1_V2](https://github.com/danomagnum/gologix/tree/master/examples/Server_Class1_V2). I edited the argument to ParsePath, but I honestly am not quite sure what these parameters mean. The PLC is in slot 0 and the ethernet module is in slot 1 (I believe, I'm not too familiar with PLC terminology). The other thing that's different from the example is that on the PLC I'm unable to create ETHERNET-BRIDGE modules, they're grayed out when I try to create a new module, so I'm attempting to use ETHERNET-MODULE. Am I doing something wrong here? Any help would be appreciated!

server.go:
```go
package main

import (
	"log"
	"os"
	"time"

	"github.com/danomagnum/gologix"
)

// these types will be the input and output data section for the io connection.
// the input/output nomenclature is from the PLC's point of view - Input goes to the PLC and output
// comes to us.
//
// the size (in bytes) of these structures has to match the size you set up in the IO tree for the IO connection.
// presumably you can also use other formats than bytes for the data type, but the sizes still have to match.
type InStr struct {
	Data  [9]byte
	Count byte
}
type OutStr struct {
	Data [10]byte
}

func main() {

	////////////////////////////////////////////////////
	// First we set up the tag providers.
	//
	// Each one will have a path and an object that fulfills the gologix.TagProvider interface
	// We set those up and then pass them to the Router object.
	// here we're using the build in io tag provider which just has 10 bytes of inputs and 10 bytes of outputs
	//
	////////////////////////////////////////////////////

	r := gologix.PathRouter{}

	// define the Input and Output instances.  (Input and output here is from the plc's perspective)
	inInstance := InStr{}
	outInstance := OutStr{}

	// an IO handler in slot 2
	//p3 := gologix.IOProvider[InStr, OutStr]{}
	p3 := gologix.IOProvider[InStr, OutStr]{
		In:  &inInstance,
		Out: &outInstance,
	}
	path3, err := gologix.ParsePath("0,1")
	if err != nil {
		log.Printf("problem parsing path. %v", err)
		os.Exit(1)
	}
	r.Handle(path3.Bytes(), &p3)

	s := gologix.NewServer(&r)
	go s.Serve()

	t := time.NewTicker(time.Second)

	for {
		<-t.C
		inInstance.Count++
		p3.InMutex.Lock()
		log.Printf("PLC Input: %v", inInstance)
		p3.InMutex.Unlock()
		p3.OutMutex.Lock()
		log.Printf("PLC Output: %v", outInstance)
		p3.OutMutex.Unlock()
	}

}
```

PLC I/O Configuration and properties of the ETHERNET-MODULE:
[image]

Server output:
```
$ go run server.go
2024/05/30 18:37:36 Listening on TCP port 44818
2024/05/30 18:37:36 Listening on UDP port 2222
2024/05/30 18:37:37 PLC Input: {[0 0 0 0 0 0 0 0 0] 1}
2024/05/30 18:37:37 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:38 PLC Input: {[0 0 0 0 0 0 0 0 0] 2}
2024/05/30 18:37:38 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:39 PLC Input: {[0 0 0 0 0 0 0 0 0] 3}
2024/05/30 18:37:39 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:40 PLC Input: {[0 0 0 0 0 0 0 0 0] 4}
2024/05/30 18:37:40 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:40 new connection from 192.168.1.11:2112
2024/05/30 18:37:40 context: 0
2024/05/30 18:37:41 PLC Input: {[0 0 0 0 0 0 0 0 0] 5}
2024/05/30 18:37:41 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:42 PLC Input: {[0 0 0 0 0 0 0 0 0] 6}
2024/05/30 18:37:42 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:43 PLC Input: {[0 0 0 0 0 0 0 0 0] 7}
2024/05/30 18:37:43 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:44 PLC Input: {[0 0 0 0 0 0 0 0 0] 8}
2024/05/30 18:37:44 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:45 PLC Input: {[0 0 0 0 0 0 0 0 0] 9}
2024/05/30 18:37:45 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:46 PLC Input: {[0 0 0 0 0 0 0 0 0] 10}
2024/05/30 18:37:46 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:47 PLC Input: {[0 0 0 0 0 0 0 0 0] 11}
2024/05/30 18:37:47 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:48 PLC Input: {[0 0 0 0 0 0 0 0 0] 12}
2024/05/30 18:37:48 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:49 PLC Input: {[0 0 0 0 0 0 0 0 0] 13}
2024/05/30 18:37:49 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:50 PLC Input: {[0 0 0 0 0 0 0 0 0] 14}
2024/05/30 18:37:50 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:51 PLC Input: {[0 0 0 0 0 0 0 0 0] 15}
2024/05/30 18:37:51 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:52 PLC Input: {[0 0 0 0 0 0 0 0 0] 16}
2024/05/30 18:37:52 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:53 PLC Input: {[0 0 0 0 0 0 0 0 0] 17}
2024/05/30 18:37:53 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:54 PLC Input: {[0 0 0 0 0 0 0 0 0] 18}
2024/05/30 18:37:54 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:55 PLC Input: {[0 0 0 0 0 0 0 0 0] 19}
2024/05/30 18:37:55 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:56 PLC Input: {[0 0 0 0 0 0 0 0 0] 20}
2024/05/30 18:37:56 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:57 Error on connnection 192.168.1.11:2112. problem reading eip header. read tcp 192.168.1.205:44818->192.168.1.11:2112: read: connection reset by peer
2024/05/30 18:37:57 PLC Input: {[0 0 0 0 0 0 0 0 0] 21}
2024/05/30 18:37:57 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:58 PLC Input: {[0 0 0 0 0 0 0 0 0] 22}
2024/05/30 18:37:58 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:37:59 PLC Input: {[0 0 0 0 0 0 0 0 0] 23}
2024/05/30 18:37:59 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:38:00 PLC Input: {[0 0 0 0 0 0 0 0 0] 24}
2024/05/30 18:38:00 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:38:01 PLC Input: {[0 0 0 0 0 0 0 0 0] 25}
2024/05/30 18:38:01 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:38:02 PLC Input: {[0 0 0 0 0 0 0 0 0] 26}
2024/05/30 18:38:02 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:38:03 PLC Input: {[0 0 0 0 0 0 0 0 0] 27}
2024/05/30 18:38:03 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:38:04 PLC Input: {[0 0 0 0 0 0 0 0 0] 28}
2024/05/30 18:38:04 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/30 18:38:05 new connection from 192.168.1.11:2113
2024/05/30 18:38:05 context: 0
^Csignal: interrupt
```

danomagnum, 2024-05-31:
What firmware are you running?  An L73 should definitely be able to add an ethernet-bridge.  It could be that the ENBT can't support it (the ENBT is quite bad performance-wise and if you can, you should get an EN2T for a lot of reasons.)

You can think of the gologix server as a virtual 1756 plc rack.  The path you pass to Handle() tells the server what data interface is available at what "slot" of the virtual rack.  In the case of the program you posted that's "0,1" which would be slot 1. ("0,1" means port 0 of the fake network card (which would be the backplane) and then slot 1. You can google "cip routing" to learn more about how that works.)

I tried on my test PLC creating a generic module (not an ethernet-bridge) 

[image]

and I got this error:

[image]

but I never got a connection reset by peer.

If I set the path to what it said ( `[]byte{52, 4}` ) and passed that as the parameter to Handle() it seems to work OK.

```
	path3 := []byte{52, 4}
	r.Handle(path3, &p3)
```

[image]

The error your getting (problem reading eip header) happens almost instantaneously after the connection is made.  Can you get a wireshark capture of it?

All this said, the server functionality is probably not super common to use - the more common way to communicate with the PLC is to poll it with a client.  Not to say that it isn't needed in your situation.

jwalke23, 2024-05-31:
[capture.zip](https://github.com/user-attachments/files/15515572/capture.zip)
I've attached a capture (note that server.go was only active for packets ~4-26). You can see the PLC (192.168.1.11) sends a single ENIP packet, but no ENIP packets are being sent from the server. It seems like the problem might lie in a problem with the UDP port, there are no packets on 2222. I've verified that I can send UDP packets to 2222 on the server, however there aren't any coming from the PLC. Using ss and netstat, I've determined that while the server is running the 2222 port is not in a LISTEN state.
```
$ sudo ss -l -n | grep -E "44818|2222"
udp                UNCONN              0                    0           *:2222                                         *:*
tcp                LISTEN              0                    4096        *:44818                                        *:*
```
I haven't found any firewall type things that should restrict this, the port works perfectly fine with other devices (using netcat).

I also have another ethernet card on the PLC, but it also has ETHERNET-BRIDGE grayed out.
[image]

Also, I verified that communication works fine polling with a client. The reason for using a server is that I'm hopeful it might solve problems with pylogix on my production PLC, which is a little overloaded and non-implicit communication gets buffered and suffers delays.

jwalke23, 2024-05-31:
Firmware and PLC configuration:
[image]
[image]
[image]

danomagnum, 2024-05-31:
Ok. I'll take a look at the packet capture this afternoon. 

The udp port doesn't come into play until the tcp handshake is done (called Forward open in cip/eip) which defines the message sizes, rpi, connection id, etc.... Then both ends start pumping udp messages at the defined rpi.  Neither side would ever know if the udp port on the opposite side wasn't open because of how udp works, but after enough missed rpi intervals the tcp connection would be closed down to let the other side know no packets were received.

From the console output you posted in the first post we're not even making it through the handshake - hopefully the packet capture will clear up what is going on.

danomagnum, 2024-05-31:
It's doing a "list services" request instead of actually opening the class 1 connection.  I'll have to see what response it is expecting and get back to you.

jwalke23, 2024-05-31:
I was just able to get it working, it turns out I wasn't able to add the bridge module because the PLC was online, and that's something you're supposed to do offline. Still doesn't explain what's going on with the generic ethernet module, if you think it's worth investigating that more, I'd be happy to help, but my issue is now resolved.

danomagnum, 2024-05-31:
I have no idea if this will work.  do a `go get github.com/danomagnum/gologix/commit/8cd66d3e3d178c6512eca4a3a715d86a45da10dd` and see if it works.  That should make it use a test version with support for that command.  My plc doesn't do that so I can't test it unfortunately.

danomagnum, 2024-05-31:
I see.  Glad it's working now.  If you wouldn't mind testing with the regular generic module and that specific commit I'd appreciate it just so I know if it works or not.

jwalke23, 2024-05-31:
Ok, I ran with that commit. I am also now using the EN3TR instead of the ENBT card (I switched to using that card earlier in addition to adding the bridge module instead). It could be that the problems were all with the ENBT card.

<details><summary>Server Output</summary>
<p>

```
2024/05/31 15:02:25 Listening on TCP port 44818
2024/05/31 15:02:25 Listening on UDP port 2222
2024/05/31 15:02:26 PLC Input: {[0 0 0 0 0 0 0 0 0] 1}
2024/05/31 15:02:26 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/31 15:02:27 PLC Input: {[0 0 0 0 0 0 0 0 0] 2}
2024/05/31 15:02:27 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/31 15:02:28 PLC Input: {[0 0 0 0 0 0 0 0 0] 3}
2024/05/31 15:02:28 PLC Output: {[0 0 0 0 0 0 0 0 0 0]}
2024/05/31 15:02:28 new connection from 192.168.1.11:51742
2024/05/31 15:02:28 context: 0
2024/05/31 15:02:28 register message: {ProtocolVersion:1 OptionFlag:0}
2024/05/31 15:02:28 context: 9223373759136661504
2024/05/31 15:02:28 ih: 0. timeout: 0
2024/05/31 15:02:28 items: [{Header:{ID:0 Length:0} Data:[] Pos:0} {Header:{ID:178 Length:60} Data:[84 2 32 6 36 1 5 155 0 0 0 0 97 64 28 0 1 0 1 0 218 184 119 0 2 0 0 0 16 39 0 0 16 72 16 39 0 0 12 72 1 9 52 4 0 0 0 0 0 0 0 0 32 4 36 102 44 101 44 100] Pos:0}]
2024/05/31 15:02:28 got small forward open from 192.168.1.11:51742
2024/05/31 15:02:28 forward open msg: {cipService_ForwardOpen 2 32 6 36 1 5 155 0 1851489 1 1 7846106 2 10000 18448 10000 18444 1 9} @ [52 4 0 0 0 0 0 0 0 0 32 4 36 102 44 101 44 100]
2024/05/31 15:02:28 New Managed Connection. {ID:1 OT:2848060441 TO:1851489 RPI:10ms Path:[52 4] Open:true}
2024/05/31 15:02:28 No tag provider for path [52 4]
2024/05/31 15:02:28 Error on connnection 192.168.1.11:51742. problem with sendrrdata problem handling forward open. no tag provider for path [52 4]
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
2024/05/31 15:02:28 couldn't find tag provider for connection 2848060441 at path [52 4] to handle IO message
[... cut ...]
2024/05/31 15:11:46 PLC Output: {[0 0 0 0 0 0 0 0 0 10]}
2024/05/31 15:11:47 PLC Input: {[0 0 0 0 0 0 0 0 0] 19}
2024/05/31 15:11:47 PLC Output: {[0 0 0 0 0 0 0 0 0 10]}
2024/05/31 15:11:48 PLC Input: {[0 0 0 0 0 0 0 0 0] 20}
2024/05/31 15:11:48 PLC Output: {[0 0 0 0 0 0 0 0 0 10]}
2024/05/31 15:11:49 PLC Input: {[0 0 0 0 0 0 0 0 0] 21}
2024/05/31 15:11:49 PLC Output: {[0 0 0 0 0 0 0 0 0 10]}
2024/05/31 15:11:50 PLC Input: {[0 0 0 0 0 0 0 0 0] 22}
2024/05/31 15:11:50 PLC Output: {[0 0 0 0 0 0 0 0 0 10]}
^C
```

</p>
</details>

danomagnum, 2024-05-31:
Thanks for the feedback.  I'm going to blame it on the enbt for now.

closed 2024-05-31 by danomagnum

danomagnum, 2024-06-01:
I was thinking about it and you might find that your communication delays you were experiencing are gone now that the enbt is replaced.

jwalke23, 2024-06-03:
Thanks, but the production PLC hasn't been using any ENBTs, all ENxTRs. The two main options I'm exploring are either getting these Linux server connections on their own comms card, because they share with other stuff, or try implicit (class 1) comms.

To illustrate some symptoms of the issue in case you have any additional thoughts, we have 3 local servers communicating with the PLC via Pylogix. The communication buffers so that all the info arrives, but from 0.3 to 1.5 seconds late. Disabling 2 of the servers, the 1 server will have nearly perfect communication, rarely more than 0.4 seconds late. However, if a second person other than me connects to the PLC then the communication of that 1 server is significantly worse, going back to being a second or even more late.

My impression is that implicit connections receive some sort of priority which may reduce the buffering that I believe is occurring. Is this a problem an implicit connection could potentially solve, or would it be made worse? From what I've heard, implicit communication can be slower, but I'm really just concerned with whether it can be consistent under these circumstances.

danomagnum, 2024-06-03:
That's interesting.  IImplicit class 1 connections will be considerably more performant than explicit class 3 connections because there isn't a poll/response - the data just gets sent at the rpi automatically and it gets a higher priority in the controller.

Have you ever used the "studio 5000 task monitor" to look at the plc communication load?  It's a handy tool for that.  The new L8's have the task monitor built in on their internal webpage, but the L7s you have to download the tool from the rockwell pcdc site.  It's free. 

[image]

[image]

 If you're using the continuous task it could be that the system overhead time slice isn't set high enough - you can adjust that from the advanced tab of the controller properties in studio 5000.

They recommend you don't use the continuous task anymore - instead using (a) periodic task(s) at a rate appropriate for the application.  Then communication tasks don't have to fight the continuous task for execution time.  Obviously that may or may not be an easy change on an existing running system.