danomagnum/gologix #20: Modification Request?

treavorj, 2024-05-31:
So not exactly an issue, but I since there isn't a Discussions area I figured this would be the next best options.

I was working on building a similar project, but then I ran across yours and it is much further developed than mine. I first started taking some inspiration from your project but then realized that I was narrowing in on similar methods for solving the problems as what you had taken so I figured I and hopefully others would be better served from just helping this project out.
Would you be interested/open to me issuing some fairly heavy updates to your code that don't change a huge amount but do make a few things more idiomatic to go and knock out some of the TODOs that were listed throughout the code?

If not, that is cool. I just didn't want to fork and issue a bunch of PRs that you are not interested in.

danomagnum, 2024-06-01:
Yeah, that would be great.  There are a number of things that I want to refactor and just haven't done.  I've purposefully kept the version in beta because I knew the api would probably change, so I'm totally open to any improvements you've got.

treavorj, 2024-06-20:
Probably will do more, but at least this was my first "go" at making some updates to the code. As I mentioned in there, please feel free to accept/reject any of the edits, and if there are some other things that you would like to see done in the code base (specifically around the client), let me know and I will see if I can take care of some of that as well.
PR: #21

closed 2024-06-28 by danomagnum

danomagnum, 2024-06-28:
I closed this, but if you'd like to make any other changes, feel free to open up more pull requests.