danomagnum/gologix #22: Clean up logging interface.

danomagnum, 2024-06-26:
Create logging interface that matches slog and provide a shim for using the regular log package.

danomagnum, 2024-06-28:
Also need to migrate "verbose" constant into probably Debug or Trace level logs and eliminate the constants file.

iceisfun, 2025-04-05:
We just started testing with this and wanted to put in our own logging when I  saw this so I just did a PR..

Do you think its acceptable to remove listalltags.go Successful member read for?

https://github.com/danomagnum/gologix/blob/f571d478af61e9ad6dfaa211d0b98749bbfd7414/listalltags.go#L229

danomagnum, 2025-04-05:
Looking at it on my phone, I'm not sure why that is an error log.  I'll be back home on Sunday and will take a look.

danomagnum, 2025-04-07:
Merged your PR and fixed the Error in listalltags.  

Beta 29.1 is released with these changes.

commit ee066168 (2025-04-07): fix #22 - error log should be debug
closed 2025-04-07 by danomagnum