danomagnum/gologix #24: read SINT8 to get a error data type is uint8

Ri0nGo, 2024-10-30:
When using readValue to read a value of type sint, the code uses byte to parse the value, but byte is essentially uint8, so the parsed data type is uint8, the correct way to parse the data is to use int8.

```go
func readValue(t CIPType, r io.Reader) (any, error) {
	var value any
	var err error
	switch t {
	......
	case CIPTypeBOOL:
		var trueval bool
		err = binary.Read(r, binary.LittleEndian, &trueval)
		value = trueval
	case CIPTypeBYTE:
                // var trueval int8
		var trueval byte   // can't use byte
		err = binary.Read(r, binary.LittleEndian, &trueval)
		value = trueval
``` 

Feel free to point out if I've misunderstood.

Thank you very much

danomagnum, 2024-10-30:
Those are good points.  I will take a look at it this evening, but nothing comes to mind as to why these fixes wouldn't work.

danomagnum, 2024-10-31:
Beta 26 now uses int8 as the correct type for SINT on the controllers.  Give it a try and let me know if it works.

I don't have one of the newer L8 controllers so I can't test against USINTs, but SINTs should work now.

Ri0nGo, 2024-11-01:
Thanks a lot, sint8 is working, I simulated a server using python as a test reference.

```python
import sys
from cpppo.server.enip.main import main as enip_main

def main( **kwds ):
    args_list = ['--print', '-v']
    args_list.append('TAG1=REAL')
    args_list.append('TAG2=BOOL')
    args_list.append('TAG3=DINT')
    args_list.append('TAG4=LINT')
    args_list.append('TAG5=INT')
    args_list.append('TAG6=SINT')
    args_list.append('TAG7=USINT')
    args_list.append('TAG8=UINT')
    args_list.append('TAG9=UDINT')
    args_list.append('TAG10=LREAL')

    # Establish Identity and TCPIP objects w/ some custom data for the test, from a config file
    return enip_main(args_list)

if __name__ == "__main__":
    sys.exit( main() )

```

Ri0nGo, 2024-11-04:
solve the problem

closed 2024-11-04 by Ri0nGo