danomagnum/gologix #26: ListAllTags recursive call causes stack overflow

songouku, 2024-12-06:
Hello, I found that ListAllTags in v0.24.0-beta recursively calls itself, causing a stack overflow. After examining the source code, I found the following code snippet. What is the status of data_hdr.Status == 6?

[image]

[image]
 
And  What should I do to avoid stack overflow?

Thanks!!!!

songouku, 2024-12-09:
I didn't specifically call the ListAllTags method; it just happened automatically after the client connection was established, as shown in the following diagram:
[image]
[image]

I checked the code and found that when KeepAliveAutoStart == true(Actually, the default KeepAliveAutoStart==true), the KeepAlive method is called, which in turn calls ListAllTags recursively. I have now set KeepAliveAutoStart to false.

danomagnum, 2024-12-09:
Yeah, the default for that was changed to false in v0.35.0-beta I think.

songouku, 2024-12-10:
OK, then I will close this issue. Thank you for your explanation!

closed 2024-12-10 by songouku