danomagnum/gologix #35: Reading Bool- & String-Arrays

led0nk, 2025-05-04:
Hey there,

I'm currently facing an issue with reading bool and string arrays.

This is my implementation for e.g. Bool:
```go
	case gologix.CIPTypeBOOL:
		arr := make([]bool, item.ArrayLength)
		err := g.CIP.Read(item.TagName, arr)
		if err != nil {
			return "", err
		}
		return fmt.Sprintf("%v", arr), nil
``` 
which works fine for the other datatypes but currently i'm getting this for the bool-array:
```
couldn't cast array. %!w(<nil>) 
```
which seemingly comes from here:
https://github.com/danomagnum/gologix/blob/f2e844a59d9441655e98a9582f2aeecb5b58451d/read.go#L444-L447

And i'm not sure why this keeps happening. The array i try to read from is implemented through the server-functionality of this library.
https://github.com/led0nk/eip-server/blob/76381addacc5743e51441686756dce904c4538b0/cmd/server/main.go#L125-L128

When reading a string-array i get this here:
```
problem reading item 2's header.
```

I also tried converting it and using as an interface-type for reading but that's obviously the same result, since both call the same case.

Do you have any suggestions to that ?

danomagnum, 2025-05-05:
In an AB PLC, all bool arrays have to be multiples of 32 bits long.

I tested against an AB controller and for a BOOL[32] array, you can't actually read it as an array - you have to read it as an atomic and the controller returns a `CIPTypeDWORD` instead of an actual bool array.

For a BOOL[64] array, you do have to read it as an array and it is returned as an array of `CIPTypeDWORD`.

I implemented additional logic in the `client.Read()` method to do this automatically for you if reading a `BOOL[32]` or multiple thereof.  It is in the newest release - [v0.30.1-beta](https://github.com/danomagnum/gologix/releases/tag/v0.30.1-beta).

Unfortunately I still don't think it will work with the server as it is today.  I'll have to implement the proper data serialization for bool and  string arrays there at some point before it will.

I also got string array reads working but there's some gotchas there that are probably there for all array types - you can't read more in one attempt than the controller's connection size allows.  This is particularly problematic for strings because they use 86 bytes + overhead each.  I'll have to add some logic to follow up on partial data transfers and finish retrieving the data.  I added an issue for that, #37

led0nk, 2025-05-05:
> For a BOOL[64] array, you do have to read it as an array and it is returned as an array of CIPTypeDWORD.
Meaning this would result in reading an Array with the length of 2 right? since the DWord is 32Bit?

I'm not sure on the ethernet/IP part of the reading data, but let's say you have an array of bool which has the length 16, i could also simply read this as an CIPTypeWord right? 
Just thinking about a kind of wrapper-solution which than can only read arrays of type bool by a factor of 8.

> I implemented additional logic in the client.Read() method to do this automatically for you if reading a BOOL[32] or multiple thereof. It is in the newest release - [v0.30.1-beta](https://github.com/danomagnum/gologix/releases/tag/v0.30.1-beta).
I'll check this out, thanks!

> I also got string array reads working but there's some gotchas there that are probably there for all array types - you can't read more in one attempt than the controller's connection size allows. [...]

Reading this, this might then result in some kind of message queue right? Reading strings is not that important for now in my case, but would be an awesome feature. Issue here is that i don't have any plc's to test with. I can only use the server-functionality of your library for that.

Thanks for the fast reply

led0nk, 2025-05-05:
>Unfortunately I still don't think it will work with the server as it is today. I'll have to implement the proper data serialization for bool and string arrays there at some point before it will.

Just to give you some review on that, i observed this when reading an array with lenght of 32 and the newest released code.
```
couldn't unpack struct header. unexpected EOF
```
And still getting this from the []string, while reading an array of length 10 of type string.
```
problem reading item 2's header. unexpected EOF
```

danomagnum, 2025-05-13:
I did confirm that if you make a smaller bool array, it rounds up to 32 and won't create an intermediate size on a rockwell controller.

I've started working on adding a client->server integration test and fixing the bool array and string array bugs.  Might take a little bit to get through it all but it's in progress.

led0nk, 2025-05-13:
okay, sounds good, let me know whenever it's done, and thanks in regards!

danomagnum, 2026-10-09:
Closing due to age.  Should be fixed.

closed 2026-10-09 by danomagnum