danomagnum/gologix #44: Issue: ReadMulti Fails When Using Array Tags

pai-bagsid, 2025-06-05:
## 🐛 Issue: `ReadMulti` Fails When Using Array Tags

Hi,

I’ve been using `gologix` to read tags from an Allen-Bradley PLC. The `Read_single()` function works fine for individual tags. However, when I attempt to use `ReadMulti()` to fetch multiple tags—particularly array tags—it fails with an `EmbeddedServiceError`.

For compliance, I've replaced the actual controller IP with `1.1.1.1`. So don't think it was wrong..
---

### 🔧 Code Example

```go
package main

import (
	"log"

	"github.com/danomagnum/gologix"
)

func main() {
	var err error
	client := gologix.NewClient("1.1.1.1")

	err = client.Connect()
	if err != nil {
		log.Printf("Error opening client: %v", err)
		return
	}
	defer client.Disconnect()

	type multiread struct {
		Tag1 []int8 `gologix:"Tag1[16]"`
		Tag2 int32  `gologix:"Tag2"`
	}

	mr := multiread{}
	mr.Tag1 = make([]int8, 16)

	err = client.ReadMulti(&mr)
	if err != nil {
		log.Printf("ReadMulti error: %v", err)
		return
	}

	log.Println("Tag1:", mr.Tag1)
	log.Println("Tag2:", mr.Tag2)
}
```

go.mod
```
module src

go 1.22.12

require (
	github.com/danomagnum/gologix v0.33.0-beta
	github.com/npat-efault/crc16 v0.0.0-20161013170008-4128ccbe47c3 // indirect
)
```

Output 
------
2025/06/05 20:20:50 INFO Session connected controllerIp=1.1.1.1 sessionHandle=1111111111
2025/06/05 20:20:50 INFO successfully opened connection controllerIp=1.1.1.1 ConnectionSize=4000 OTNetworkConnectionId=1111111111
**2025/06/05 20:20:50 ReadMulti error problem in read list: service returned status EmbeddedServiceError**
2025/06/05 20:20:50 INFO starting disconnection controllerIp=1.1.1.1
2025/06/05 20:20:50 INFO successfully disconnected from controller controllerIp=1.1.1.1

Diagnosis and Fix
------------------------
After some investigation, I found that the issue stemmed from uninitialized array fields. I made changes to support array tag reads properly, and it worked.

I also attempted to run your test suite but could not due to missing JSON files.

To help, I’ve submitted a PR with the proposed fix:
🔗 [PR #43 – Fix ReadMulti for Array Tags](https://github.com/danomagnum/gologix/pull/43)

Fix and Update
--------------------
To test updated the go.mod and ed below line to point to my repo for the fix
```
replace github.com/danomagnum/gologix => github.com/pai-bagsid/gologix multireadtag-fix
```

After Fix output
---------------------
2025/06/05 20:32:06 INFO Session connected controllerIp=1.1.1.1 sessionHandle=1111111111
2025/06/05 20:32:06 INFO successfully opened connection controllerIp=1.1.1.1 ConnectionSize=4000 OTNetworkConnectionId=1111111111
2025/06/05 20:32:06 Tag1 : [23 54 33 33 34 50 34 50 50 50 40 30 0 0 0 0]
2025/06/05 20:32:06 Tag2 : 23381
2025/06/05 20:32:06 INFO starting disconnection controllerIp=1.1.1.1
2025/06/05 20:32:06 INFO successfully disconnected from controller controllerIp=1.1.1.1

danomagnum, 2025-06-05:
Makes sense to me just glancing at it. I would expect the array index on the struct tag to be the starting index from the array to read and the length of the slice to indicate how many elements to read so that would mean  you read  Tag1[16] through Tag1[31]?  

I'll try to get the tests verified this evening.

(Note that if you wanted to run the tests you would copy the [test_config template.json](https://github.com/danomagnum/gologix/blob/master/tests/test_config%20template.json) file as `test_config.json`, load the [gologix_tests_Program.L5X](https://github.com/danomagnum/gologix/blob/master/tests/gologix_tests_Program.L5X) as a program into the controller you want to test with, and then configure the json file with that controllers info.  I'll add a readme.md file to the tests directory about that when I get a chance.)

danomagnum, 2025-06-06:
I did some looking and you should already be able to read arrays in a multi read.  The gologix tag should match the first element you want to read and the size of the slice should indicate the number of elements.  

See the test here: https://github.com/danomagnum/gologix/blob/93e61abcb3d3ae18d5381d9a3e3b3494f45ad399/tests/read_multi_test.go#L162 where it reads 5 elements from `TestDintArr` starting at element 2.  I also tested that if you want to read starting at element 0 you can either use `TestDintArr[0]` or omit the array index altogether `TestDintArr`.

So in your example I think you would want 
```go
type multiread struct {
	Tag1 []int8 `gologix:"Tag1"`
	Tag2 int32  `gologix:"Tag2[0]"`
}
mr := multiread{}
mr.Tag1 = make([]int8, 16)

```

If that doesn't work, what controller / firmware are you using?

As I was testing I fixed readList to support read requests that are larger than a single packet.  I went ahead and fixed readMulti to use this new function since that is where I've been migrating all the reads. That should now be merged and the active release.

danomagnum, 2025-06-06:
> Acutally it was like TestDintArr is a tag with 7 elements and you wanted to read from 2 to 7 correct ?

That's right.

I ran your example more or less verbatim and I did find a bug where []int8 wasn't being handled properly.  I got that fixed in beta 34.1.  I don't think that would cause the error you were getting though - it was causing a panic on the reflection to update the struct field.

After the fix and with the following program:
```go
package main

import (
	"fmt"

	"github.com/danomagnum/gologix"
)

func main() {

	client := gologix.NewClient("1.1.1.1")
	err := client.Connect()
	if err != nil {
		panic(err)
	}
	defer client.Disconnect()
	type multiread struct {
		Tag1 []int8 `gologix:"Tag1"`
		Tag2 int32  `gologix:"Tag2"` 
	}
	mr := multiread{}
	mr.Tag1 = make([]int8, 16)

	err = client.ReadMulti(&mr)
	if err != nil {
		panic(err)
	}
	fmt.Printf("Value: %+v", mr)

}

```
and this tag setup:

[image]

I get the following:
```
2025/06/06 08:42:47 INFO Session connected controllerIp=1.1.1.1 sessionHandle=1073741907
2025/06/06 08:42:47 INFO successfully opened connection controllerIp=1.1.1.1 ConnectionSize=4000 OTNetworkConnectionId=4294787198
Value: {Tag1:[100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115] Tag2:123456}
2025/06/06 08:42:54 INFO starting disconnection controllerIp=1.1.1.1
2025/06/06 08:42:54 INFO successfully disconnected from controller controllerIp=1.1.1.1
```

Your struct had Tag2 as an int32 but you also said it was a SINT.   You've also got the 0 array index on Tag2 instead of Tag1.

pai-bagsid, 2025-06-21:
Thank you for testing and poiting out the mistake

The Tag1 is  "SINT: Signed 8-bit integer value" and 
Tag2 is "DINT: Signed 32-bit integer value".

Now there is no error when running but I am not getting any value.
While I run using 'single_read' I am able to get values.

I am using the latest package still not getting value

```
package main

import (
        "log"

        "github.com/danomagnum/gologix"
)

func main() {
        var err error
        client := gologix.NewClient("1.1.1.1")

        // connect using parameters in the client struct
        err = client.Connect()
        if err != nil {
                log.Printf("Error opening client. %v", err)
                return
        }
        defer client.Disconnect()
        type multiread struct {
                Tag1 []int8 `gologix: "Tag1"`
                Tag2 int32 `gologix: "Tag2"`
        }

        mr := multiread{}
        mr.Tag1 = make([]int8, 16)
        err = client.ReadMulti(&mr)
        if err != nil {
                log.Printf("ReadMulti error %v", err)
                return
        }

        log.Println("Tag1 :", mr.Tag1)
        log.Println("Tag2 :", mr.Tag2)
}
```

Output
```
go: downloading github.com/npat-efault/crc16 v0.0.0-20161013170008-4128ccbe47c3
2025/06/21 13:54:42 INFO Session connected controllerIp=1.1.1.1 sessionHandle=11111
2025/06/21 13:54:42 INFO successfully opened connection controllerIp=1.1.1.1 ConnectionSize=4000 OTNetworkConnectionId=111111
2025/06/21 13:54:42 Tag1 : [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
2025/06/21 13:54:42 Tag2 : 0
2025/06/21 13:54:42 INFO starting disconnection controllerIp=1.1.1.1
2025/06/21 13:54:42 INFO successfully disconnected from controller controllerIp=1.1.1.1

```

Single_read 
```
package main

import (
        "log"

        "github.com/danomagnum/gologix"
)

func main() {
        var err error
        client := gologix.NewClient("1.1.1.1")

        // connect using parameters in the client struct
        err = client.Connect()
        if err != nil {
                log.Printf("Error opening client. %v", err)
                return
        }
        defer client.Disconnect()

        qty := uint16(16)
        tag1, err := client.Read_single("Tag1", gologix.CIPTypeSINT, qty)
        if err != nil {
                log.Printf("Error single read %v", err)
                return
        }
        log.Println("Tag1 : ", tag1)

        tag2, err := client.Read_single("Tag2", gologix.CIPTypeDINT, 1)
        if err != nil {
                log.Printf("Error single read %v", err)
                return
        }
        log.Println("Tag2 : ", tag2)
}
```

Output
```
go: downloading github.com/danomagnum/gologix v0.34.0-beta
go: downloading github.com/npat-efault/crc16 v0.0.0-20161013170008-4128ccbe47c3
2025/06/21 13:56:31 INFO Session connected controllerIp=1.1.1.1 sessionHandle=1111
2025/06/21 13:56:31 INFO successfully opened connection controllerIp=1.1.1.1 ConnectionSize=4000 OTNetworkConnectionId=11111
2025/06/21 13:56:31 Tag1 :  [16 41 19 12 26 26 31 32 54 48 20 20 30 40 50 60]
2025/06/21 13:56:31 Tag2 :  8
2025/06/21 13:56:31 INFO starting disconnection controllerIp=1.1.1.1
2025/06/21 13:56:31 INFO successfully disconnected from controller controllerIp=1.1.1.1

```