danomagnum/gologix #47: Device cannot be disconnected

Ri0nGo, 2025-07-15:
Hello, my program implements a set of automatic reconnect logic (autoconnect of gologix is turned off), and occasionally the device cannot be disconnected. Have you ever encountered such a situation?

danomagnum, 2025-07-15:
That doesn't sound familiar.  Do you get an error or does it just lock up? 

Some devices have a maximum connection count which can be surprisingly low.  If you're losing connection it's possible the old connection is still "open" on the device end.  Usually they time out after a few minutes and you can connect again.  If I remember you get an error on the connection attempt when this happens though.

danomagnum, 2025-07-15:
Have you overridden the client's `SocketTimeout` property?  It looks like it defaults to 10 seconds, but if it is zero it uses 1 second instead.  That should prevent any blocking from network errors for longer than whatever the timeout is.

Do you have a debug level logger set up?  That may be helpful.

Ri0nGo, 2025-07-29:
Hi, I find a dead lock when connect device

[image]

Tests found that there was indeed this problem

[image]

danomagnum, 2025-07-30:
That makes sense to me.

I believe the changes in the https://github.com/danomagnum/gologix/tree/DisconnectDeadlock branch will take care of this by more rigorously handling the connection state.  Now if there is a connection issue when connecting, it will abort the disconnect.  When the error bubbles back to the connect routine, it will set everything back to disconnected.

Give that branch a try and let me know if it works for you.

Do you have a minimum code example that reproduces this behavior which I could use to add a test?

Ri0nGo, 2025-07-30:
I haven't tested the code for the DisconnectDeadlock branch yet, I will let you know if there is any result

minimum code example:
1. Launch an available server, Modify the `sendData` function in the `sendreceive.go` file and proactively throw an error,
```go
func (client *Client) sendData(b []byte) error {
	// write the packet buffer to the tcp connection
	written := 0
	for written < len(b) {
	    ........
	}
	return fmt.Errorf("mock a error")  // return a error
	//return nil

}
```

2. Add a log， `disconnect.go/Disconnect()` 
```go
func (client *Client) Disconnect() error {
	if client.connecting {
		client.Logger.Debug("waiting for client to finish connecting before disconnecting")
		for client.connecting {
			time.Sleep(time.Second * 2)
			client.Logger.Info("sleep 2 second")  // add log
		}
	}
    ......
}
```

3. This is the test program, fmt.Println("connect complete!") will not be executed, The console keeps printing "sleep 2 second"
```go
func main() {
	ip := "192.168.1.108"
	client := gologix.NewClient(ip)
	err := client.Connect()
	if err != nil {
		fmt.Println("connect err:", err)
	}
	fmt.Println("connect complete!")
	time.Sleep(time.Second * 1000)
}
```

[image]

4. If you call the Disconnect method at this time, you will also enter the for loop of the Disconnect method and keep blocking

danomagnum, 2025-08-12:
I was finally able to get around to trying your suggested test method.  Looked like it worked to me and all the tests still pass so I'm going to merge the changes into main.

closed 2025-12-02 by danomagnum